betaflight/betaflight-configurator #2384: Can´t install 10.8.0 configurator on Windows 7 64bits machine.

nangel0, 2021-01-12:
Had to re-install Windows. Also needed to update the configurator so went straight to 10.8.0 and I can't install the last version, nor the nightly.

If I try to install the previous version 10.7.0 it works ok like before, nothing changed on hardware.

[image]
[image]
[image]

Asizon, 2021-01-12:
Hello! Windows 7 is no longer supported by Google and also by Betaflight Configurator.

nangel0, 2021-01-12:
Face-palm... Not aware off that... Time to upgrade OS...

closed 2021-01-12 by nangel0

Droneman72, 2022-05-13:
Nothing wrong with BF 10.7.2 anyway, and I am not about to replace my Win7 Ultimate in my main PC because Google and BF do not support it.  I just loaded it in a cheap Win10 notebook, which is also light and portable if needed at a flying site. Where there is a will there is a way!