betaflight/betaflight-configurator #2385: Firmware flash on 10.7 fails on SPRACINGF3

dribbe, 2021-01-13:
Hi,
Seems like a BUG (10.7 not compatible).
When attempting to flash Firmware using BF Configurator 10.7 (Win10 PC), the flash progresses then 'fails' with RED BAR about 10% or about 50% into the verification process (code appears new/flashed OK after power cycle and re-connect. ) Same code and PCB hardware and cable works fine on 10.6.
Does it every time.

Duplicate with 10.7, SPRACINGF3 hardware, and the prebuilt 4.0.6 file in menu at top selecting SPRACINGF3.
(have duplicated with other FW release versions also.)
Have tried different F3 PCB's, Cables, Baud Rates, PC's, and FW files.... 10.7 always fails. 10.6 or 10.5 always work with all variations present.
(also tried normal flash, boot pin flash, full erase, etc... makes no difference. 10.7 fails always... others work always.)

Best Regards,
David

danybd, 2021-03-10:
Hi,
I want to confirm the issue as well. I tried 2 days ago to update my flight controller (SPracingF3) and almost freak out. At the end I did the update using ST toolchain + ST link. I´ve got the same setup as described by dribbe above.

haslinghuis, 2021-03-10:
With an unsupported FC (BrainFPV Radix) I have the exact same problem, the progress bar stops at 50%. But don't freak out. Wait a little and power cycle (toggle USB) and it works. Not sure about F3. We only support F4 and up and that's where I started :smile:

closed 2022-11-05 by haslinghuis