betaflight/betaflight-configurator #2393: last master overwrites wrong config data

TheIsotopes, 2021-01-21:
since #1821 a wrong config value is overwritten after setting the sliders.

@IvoFPV pls can you check this.

```
# get curve
dyn_lpf_gyro_curve_expo = 256
###ERROR IN get: CORRUPTED CONFIG: dyn_lpf_gyro_curve_expo = 256###
Allowed range: 0 - 10
Default value: 0
```

diff output:
```
...
# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_curve_expo = 256
###ERROR IN diff: CORRUPTED CONFIG: dyn_lpf_gyro_curve_expo = 256###
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = FPORT
...
```

haslinghuis, 2021-01-23:
`dyn_lpf_curve_expo` and `dyn_lpf_gyro_expo`. Tried to exchange, setting defaults, but not enough to restore sliders. :confused:

mikeller, 2021-01-24:
@TheIsotopes: Can you please add steps for how this can be reproduced?

TheIsotopes, 2021-01-24:
@mikeller done ... see EDIT2 above

Asizon, 2021-01-24:
Ok, so the problem seems that dyn_lpf_gyro_curve is part of simplified gyro filter but this feature is not yet added to MSP and Configurator so its breaking the value.

IllusionFpv, 2021-01-24:
Maybe this can be reopened https://github.com/betaflight/betaflight/pull/10240

TheIsotopes, 2021-01-24:
@IllusionFpv ok, tested now and the first bug is fixed. 👍 

@all now the problem is still missing that the slider does not appear in the filter settings.

haslinghuis, 2021-02-01:
@TheIsotopes - found the problem. Please try again.

haslinghuis, 2021-02-07:
Found another problem, We need to fix the tuning sliders for firmware below API 1.44. Now they are disabled by default but can be enabled but don't load after save - re-enter tab.

Didn't publish a fix yet but can load and save now with sliders enabled in legacy mode, but using the sliders is not working yet. Since sliders have many changes it's hard back-tracking. Running in circles at the moment.

PR #2401 (2021-02-09): Fix firmware filter tuning sliders
Fixes #2393.
closed 2021-02-08 by mikeller