betaflight/betaflight-configurator #2397: Frozen motor sliders

salukikev, 2021-01-25:
I'm having consistent issues with the motor tab when configured as a hex.  This is being discussed here:
https://intofpv.com/t-new-to-remap-almost-got-it-but-can-t-find-a-reliable-resource-for-m6-help?pid=127283#pid127283
with user SnowleopardFPV a much more experienced user noting that this is a problem with betaflight/firmware.  Two problems are evident:
1. Only master slider is moveable, but produces no result other than being moveable.
2. The Hex configure choice doesn't configure correctly from the pulldown.  Evidently you need to add configuration lines in CLI:
mmix reset
mmix load HEX6X

These may or may not be related to a possible hardware issue that is preventing all motors from working correctly.  Only a remapped LED_STRIP resource correctly spins a motor.  No other resource that I've found on two different FC's allow the 6th motor to spin- but rather to just jitter in response to throttle.  Despite this- the remapped LED_STRIP resource is persistently successful at spinning a motor.
[Video of frozen sliders](https://i.imgur.com/TXBDxPe.mp4) (sorry about invisible cursor- I'm trying to move them but the cursor wasn't captured)
[Video of motor jitter.](https://i.imgur.com/qIDasQr.mp4)

Asizon, 2021-01-25:
@salukikev can you please paste a diff from cli here?

salukikev, 2021-01-25:
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: JBF7, version: caed8ab7, date: 2020-08-19T12:55:56Z

# start the command batch
batch start

board_name JBF7
manufacturer_id IFRC

# resources
resource MOTOR 5 A01
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_RX 4 NONE

# mixer
mixer HEX6X
mmix 0  1.000 -0.500  0.866  1.000
mmix 1  1.000 -0.500 -0.866  1.000
mmix 2  1.000  0.500  0.866 -1.000
mmix 3  1.000  0.500 -0.866 -1.000
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

# aux
aux 0 0 0 900 1225 0 0
aux 1 27 1 900 1200 0 0
aux 2 27 0 900 900 0 36

# master
set acc_calibration = -84,24,-41,1
set motor_pwm_protocol = DSHOT300
set motor_poles = 12

profile 0

rateprofile 0

# end the command batch
batch end

#

salukikev, 2021-01-25:
**I know this shows 5 active motors because I'm mid-experiment right now, but it doesn't matter.  This problem exists regardless of motor assignments.  It exists right now.

haslinghuis, 2021-01-25:
Tested on matek f722 mini - hex models work, octo does not have enough motor outputs (assigned resources with hardware timer).

haslinghuis, 2021-01-26:
@salukikev you need at least the motor resources assigned you are requesting in your mix.

haslinghuis, 2021-01-26:
Thanks for the reminder to work on this https://github.com/betaflight/betaflight-configurator/issues/2360

salukikev, 2021-01-26:
I posted a version also at #slack that has all the motors accounted for- same behavior.  I'm not experienced enough to offer much insight as to why, but the same behavior was observed from another user in the UK after taking the same configurations steps.
That said, I gave up on my FC and went back to a NAZA for now.  I have a Matek on the way so hopefully that will put this symptom to bed for me.

haslinghuis, 2021-01-26:
@salukikev it does matter to have the minimal required outputs (motor resources) assigned otherwise all outputs are 0. How many outputs are visible when using blheli configurator ?

salukikev, 2021-01-26:
I haven't yet used the blheli configurator- but the outputs are correct now.  I think you're saying that the slider freeze behavior is an intentional result indicating a mismatch of assigned motor outputs to programmed motor outputs?  I'm pretty sure other examples (per slack diff) had the proper # of motors defined, but for now I'm using a NAZA until my Matek arrives- then I plan to go back to Betaflight.  Side note: I also found a suspiciously soldered wire on the ESC of motor 6.  Although I did swap wires to verify it wasn't a motor/esc problem this might have been intermittent.  I sure hope that was the problem.

closed 2021-03-05 by github-actions[bot]