betaflight/betaflight-configurator #2398: [Firmware Support] Add support for OSD element type variants

etracer65, 2021-01-27:
Related to https://github.com/betaflight/betaflight/pull/10517

The previously unused 2 high bits of the OSD item position now contain an indicator of what optional "variant" of the element is selected. So there can be up to 4 variants of an element optionally selected by the user. The default remains unchanged and it will be the `00` option. The purpose is to have elements that might display differently based on user preference.

As noted in the related PR the "Battery usage" element has had different type support added in the firmware as an initial example.

So the goals this issue are:
1. add support for handling the extra bits int he item position to read/set the selected variant properly
2. add a user interface to display/select the variant options
3. have the preview properly display the selected variant
4. make it (hopefully) easy to add variants to other elements in the future

For item 2 I was thinking about having a drop-down next to the profile checkboxes for any element that has more than the default variant option. But that's just a suggestion.

etracer65, 2021-02-15:
Added variants for the Altitude OSD element. See https://github.com/betaflight/betaflight/pull/10551

etracer65, 2021-02-15:
I'm not sure if anybody is available to work on this. We have the opportunity to add quite a few OSD element variants to the firmware and without adding support to the Configurator users won't get much benefit. Manipulating the 2 high bits in the elements position setting in the CLI won't be very user friendly.

I've done some small stuff in the Configurator but it's not really my area of expertise and I feel like I'd struggle disproportionally trying to make this work in a usable way. @McGiverGim, @haslinghuis, @Asizon or @chmelevskij - want to give it a try?

My thought was somehow define the variants for individual elements using constant arrays for easy extension in the future. Then for the presentation I was thinking about having a drop-down for the variants next to the profile checkboxes for each element. The drop-down would only appear if the element has more than the default variant. The other complication is that the element preview should also be related to the variant. Probably also have to somehow handle cases where the firmware specified a variant that the Configurator isn't aware of yet. Something like how we display "Unknown element" then the Configurator doesn't know about an element yet.

etracer65, 2021-02-18:
If nobody has the time or desire to work on this user interface, then at the least we need to make small changes to preserve the two high bits in the item position. Currently the Configurator will zero-out these bits if the element is moved resulting in the default (type 1) variant being selected while overriding the user setting (made in the CLI).

etracer65, 2021-02-18:
Added support for GPS coordinate display variants. See https://github.com/betaflight/betaflight/pull/10566

haslinghuis, 2021-02-20:
@etracer65 will look at it, occupied with other stuff. I know it would not be easy.

McGiverGim, 2021-04-05:
I think I've some time, I will look into this...

closed 2021-06-12 by blckmn