betaflight/betaflight-configurator #2404: OSD vs OSD with DJI

zigomatichub, 2021-02-04:
As a Betaflight FPV user, it would be nice to see in OSD Tab of the configurator if an OSD item are usable with DJI goggle.

Then next to the name of this OSD item, having a "*" can refer to DJI compatibility or any kind of feature that can easily select compatible DJI osd items.

Thank you and I'm already very happy to use BetaFlight :)

haslinghuis, 2021-02-04:
Afaik DJI is not collaborating with Betaflight, so this would not be maintainable.

zigomatichub, 2021-02-05:
Up to the betaflight team to decide if it's maintainable or too much work for implementation.  
You may found a correct way to implement via an updated custom file that give support list of OSD items that configurator can read and give a visual flag on the screen to identify which items are compatible. 
May be DJI will change is mind for future, we never know

etracer65, 2021-02-06:
@zigomatichub As @haslinghuis stated, DJI doesn't collaborate with us. Betaflight has no visibility into which elements DJI supports. Further if DJI makes a firmware update and adds or removes elements there is no way for us to know what version the user might be using and what elements are or aren't available. Lastly, there is no way that we will be releasing Configurator updates just to support changes in DJI's firmware.

Your better course of action is tell DJI that that they should implement their OSD properly using MSP displayPort as provided by Betaflight so that **ALL** elements and the OSD menus will work.

zigomatichub, 2021-02-09:
Understood and good to give the clarity on why DJI should focus on. How to discuss with DJI on this ?  This is another story :)

closed 2021-03-18 by zigomatichub