betaflight/betaflight-configurator #2405: So ive been using betflight 10.7.0 for a few months now and has worked great. however this weekend i went open it and betflight was in a different language!?! like Russian or something. anyone know how to get it back to English? plz.

2stixs, 2021-02-07:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

haslinghuis, 2021-02-07:
Open configurator and look ate the bottom of the screen where you can click your language.

[image]

closed 2021-02-07 by mikeller

kolyfan, 2024-09-13:
Spend some time to find this place from screenshot. Just small notice - this options for change language available only on Welcome screen, before you connect to flight controller.