betaflight/betaflight-configurator #2406: Configurator does not load hex files

TheIsotopes, 2021-02-08:
since latest build the configurator does not load any hex files.

[image]

imo it comes from #2368 ... @chmelevskij pls can you fix this

mikeller, 2021-02-08:
@TheIsotopes: Thanks for reporting.

@chmelevskij: Uh oh, I saw that one, checked it, and thought it must be unused as well. :-(

chmelevskij, 2021-02-08:
Hmm, yeah, so workers use different execution context, hence the missing thing which was attached to global. I'm surprised it worked before though 🤔

mikeller, 2021-02-08:
You removed one that was defined inside the worker. ;-)

chmelevskij, 2021-02-08:
Ah, ok, my bad 😆  makes sense now.

McGiverGim, 2021-02-08:
Maybe for this cases that we only want to log some elapsed time is better to use `console.log()` or `console.timeLog()`. They are just there for that.

PR #2408 (2021-02-14): feat: use `console.time` for timing
Fixes #2406 

@McGiverGim used `console.time` stuff. 

@TheIsotopes could you verify it works?
closed 2021-02-14 by mikeller