betaflight/betaflight-configurator #2410: BF Config suddenly stops opening/ taskbar hover preview is blank

LAWMAN-FPV, 2021-02-14:
BF Configurator 10.7 was working for months and suddenly stopped. Program won't load beyond placing icon in task bar. 
No hardware changes. May have been a windows update? 

Tried reinstall and the --disable-gpu from a few years back but that did not work either. 
Please advise on what else I should place here to diagnose. 
Thanks and sorry if I offended any rules. First timer here.
[image]

mikeller, 2021-02-14:
It would help if you specified what operating system you are using.

LAWMAN-FPV, 2021-02-14:
OS Name	Microsoft Windows 10 Pro
Version	10.0.19042 Build 19042

mikeller, 2021-02-14:
@LAWMAN-FPV: Try uninstalling configurator, then deleting the `C:\Users\${USERNAME})\AppData\Local\betaflight-configurator` directory, and then installing it again.

LAWMAN-FPV, 2021-02-14:
That seems to have solved the issue. Thank you!

mikeller, 2021-02-15:
Nothing much to explain - this is the directory where the framework used for configurator stores its configuration and cache. Sometimes this data gets corrupted, and deleting the directory will fix this. (The directory is not deleted during an uninstall / reinstall so that the user's preferences will not be lost in every upgrade.)

haslinghuis, 2021-03-07:
@LAWMAN-FPV Please close the issue as it's resolved.

closed 2021-03-07 by LAWMAN-FPV