betaflight/betaflight-configurator #2412: Firmware sliders doesnt work

Asizon, 2021-02-14:
After https://github.com/betaflight/betaflight-configurator/pull/2409 firmware sliders doesnt work for 4.3.
To reproduce, go to pidtuning tab, check RPY and move sliders.
Using latest 4.3 and latest Configurator master build.
[image]

klutvott123, 2021-02-14:
Can confirm. And if I move a slider and click "refresh" I get this:
[image]

haslinghuis, 2021-02-14:
Changes made only relate to API 1.43 or below, so not sure how this is able to affect 1.44 behavior at all. :confused:

PR #2415 (2021-02-16): Fix Legacy Tuning Slider Patch
Fixes #2412 after #2409 (Thanks @klutvott123 for spotting the bug)

For `legacy mode` only this patch also addresses:

- [x] Changed all `floor` functions to `round` as it was before.
- [x] Hide new firmware sliders.
- [x] Added back Dmin Switch change() function.
- [x] Remove `sliderPidsModeSelect`
closed 2021-02-16 by mikeller