betaflight/betaflight-configurator #2413: New motor tab Save problems

Asizon, 2021-02-14:
With the new motor tab save system, if user wrong check some option and then uncheck it, motor test, motor reorder and motor direction features doesnt work until you saved a non changed configuration.
To reproduce:
-Check some option and the uncheck
-See how features disappears
[image]

haslinghuis, 2021-02-14:
This is by design. Change a setting and motor tests are disabled. To improve we could keep the state to avoid having to use Save and Reboot if nothing actual changed (after reverting a change).

Asizon, 2021-02-17:
Yes, i think if user reverts a change no need to reboot FC, it can be confusing with no changes:
[image]

Asizon, 2021-02-21:
@haslinghuis Also note, that if you change to another FC, save problems persists in this fc from tthe previous fc and need to aplies non made changes in this FC.

haslinghuis, 2021-02-23:
`TABS.motors.isDirty` is initialized as false on entering the Motor tabs but it's retaining its value - after re-entering tab / reboot of FC? Restarting BFC does work. Resetting the isDirty flags to false in TABS.motors.initialize function solves this.

The question of changing a value and setting it back to initial will still have the isDirty flag raised. @chmelevskij does Vue has some state management dealing with this issue? 

https://css-tricks.com/build-a-state-management-system-with-vanilla-javascript/

chmelevskij, 2021-02-23:
Vue js or just plain js, it's a matter of dealing with changes properly IMO. This can be done with or without it.

It would be nice to move it to Vue, but in current state it's hard to move complex components. I've been sticking to tidy up modules and simple components for now. There is a quite a lot of global variables and actions going on, so it's hard to track these things.

haslinghuis, 2021-03-05:
Also fixed some UI design issues, the number of motors detected (still fails if a user select octo while having only 6 outputs (resources assigned). Had to change some variables because they were no valid key-pair key names. Fixed the flip-flopping of buttons by disabling where applicable.

haslinghuis, 2021-03-06:
@Asizon all should be fixed now.

- Fixed persisent behavior when changing settings. Further fixed some UI design issues (should look better on 1366 and 1920).
- Had to change some variables because they were no valid key-pair key names. 
- Fixed the flip-flopping of buttons by disabling where applicable.
- The number of motors detected is coming from models now (but still fails if a user select octo while having only 6 outputs (resources assigned)).

PR #2428 (2021-03-22): Fix persistent settings
Fixes: #2413 

- [x] Fixed persisent behavior when changing settings including the flip-flopping of buttons by disabling where applicable.

Moved to separate PRs:
- [x] The number of motors detected is coming from models now. (#2436)
- [x] Fixed some UI design issues - should look better on 1366 and 1920 (#2448)
- [x] Fixed duplicate MSP calls at loading (#2449)

This PR is replacing the buttons with a new set. Had to rename some variables to be used properly (i.e. variables starting with a digit or containing a hyphen were not allowed). The code is defining default con [... cut]
closed 2021-03-22 by mikeller