betaflight/betaflight-configurator #2419: The firmware downloaded from the firmware update page in the Betaflight Configuration  is confused

Xiang-FPV, 2021-02-17:
### Describe the bug

In the firmware download interface, version 4.2.8 shows stm32f405. In fact, it downloads stm32f411. The display shows stm32f411. In fact, the download is stm32f405. The two firmware have been exchanged

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

mikeller, 2021-02-17:
Just tested this and it is definitely downloading / installing STM32F411 when you select it:

[image]

[image]

Xiang-FPV, 2021-02-17:
https://user-images.githubusercontent.com/68484127/108184331-07f3ce00-7146-11eb-9529-176fe247b78e.mp4

https://user-images.githubusercontent.com/68484127/108184500-4093a780-7146-11eb-87aa-07a22dca7edf.mp4

mikeller, 2021-02-17:
@Xiang-FPV: This looks weird. Have you actually checked that the firmware on your flight controller is updated, or could it be that it just is not flashed at all?
Are there any other devices connected to your computer that could be flashed? For example the Jumper series TX register as STM DFU devices when they are charging over USB.

Xiang-FPV, 2021-02-17:
At present, there is no other computer that can be used for testing. Let me send a video of the whole process of updating the firmware first

https://user-images.githubusercontent.com/68484127/108194456-a0438000-7151-11eb-8a54-6014139d6c5b.mp4

mikeller, 2021-02-17:
This is really weird. But it could be the result of cache corruption. Can you please try uninstalling configurator, then deleting the `C:\Users\${USERNAME})\AppData\Local\betaflight-configurator` directory, and then installing it again, and try flashing again?

Xiang-FPV, 2021-02-17:
OK, I downloaded it again. The problem has been solved and the corresponding firmware has become normal. Thank you very much for helping me!

haslinghuis, 2021-03-07:
@Xiang-FPV please close the issue as it's resolved.

closed 2021-03-07 by McGiverGim