betaflight/betaflight-configurator #2420: MacOs: dark mode has dark descriptions too

AndreaCCIE, 2021-02-17:
the latest Nightly builds (v20210217.1990, v20210216.1974 and v20210216.1971)have a wrong description colour:

[image]

haslinghuis, 2021-02-17:
Thanks. @Asizon can you please take a look at how to solve the font color for the descriptions in dark mode?

haslinghuis, 2021-02-17:
Fix is ready, also for servo tab:

[image]

haslinghuis, 2021-02-17:
@Asizon can't find why the line is broken in:

[image]

Asizon, 2021-02-18:
@haslinghuis for the broken line, maybe "border-colapse" can help.

haslinghuis, 2021-02-18:
Thanks @Asizon - that solved it!

PR #2421 (2021-02-21): Fix darkmode on motor + servo tab + messages
Fixes: #2420
Fixes: #2422
closed 2021-02-21 by mikeller