betaflight/betaflight-configurator #2423: @nangel0 , make sure you are on the latest Master OR 4.2.  Below is the behavior I'm getting after flashing back and forth from Master to 4.2.8.

nangel0, 2021-02-18:
@nangel0 , make sure you are on the latest Master OR 4.2.  Below is the behavior I'm getting after flashing back and forth from Master to 4.2.8.

Note: even the same target.

[image]

_Originally posted by @spatzengr in https://github.com/betaflight/betaflight-configurator/issues/2402#issuecomment-781322015_

Thanks for the tip, but still not working, if I fash 4.2.8 from 15/02/2021 it seems ok, usual sliders change values, and the refresh button brings them back to the center if not saved, and the save button works as expected.

https://www.dropbox.com/s/guech8mk5m0pznc/4.2.8.mp4?dl=0

Flashing 4.3.0 from 18/02/2021 seems to change values but the refresh button doesn't bring to the latest values and doesn't save some changes...

https://www.dropbox.com/s/92ekl0n0uxiduqv/4.3.0.mp4?dl=0

nangel0, 2021-02-18:
please delete

closed 2021-02-18 by nangel0

spatzengr, 2021-02-18:
loose USB?