betaflight/betaflight-configurator #2425: No Save and Reboot Button with Certain Screen Resolutions

akira7799, 2021-02-21:
Within Betaflight Configurator v10.7.0, the "Save and Reboot" button does not appear on a Windows 10 Laptop at 1366 x 
768 Screen Resolution, at 100% magnification.  Upon changing to 125%, the "Save and Reboot" button re-apears, however, workflow is decreased at higher magnification levels and viewability of Betaflight Configurator is decreased.

Hardware Information
Acer Laptop
AMD Ryzen 3 2200U  
Radeon Vega Mobile GFx 2.50Ghz
8GB RAM

OS Information:
Windows 10 Home Version 2004 
Build 19041.804

haslinghuis, 2021-02-21:
Is this the same on Ports, Configuration and Failsafe tab? Is the window maximized?

mikeller, 2021-02-21:
@akira7799: Can you please supply some screenshots of how the configurator window looks when the button does not appear, and how it looks when it does?

haslinghuis, 2021-03-07:
@akira7799 As we can't reproduce and there are no other reports can we close this issue?

akira7799, 2021-03-07:
Absolutely.  Thanks for looking into it.

haslinghuis, 2021-03-16:
@McGiverGim please close

closed 2021-03-16 by McGiverGim