betaflight/betaflight-configurator #2426: BF Configurator wont launch?

huge-11, 2021-02-22:
Hello,
i installed version 10.7.0  but somehow i cant launch the configurator. When i do so just nothing happens ? 
I am on win 10 home
version 1909
build 18363.1379

tried to uninstall and reinstall but still same. 
i have no folder in ...AppData/Local/ called betaflight so i cant delete it like in #2410

any help would be apreciated

haslinghuis, 2021-02-22:
Could you run compatibility troubleshooter?

[image]

Also see https://github.com/betaflight/betaflight-configurator#graphics-issues

huge-11, 2021-02-22:
So i uninstalled bf config with revo uninstaller, removed all the registrys and everything related to br config. Did some gpu config and updated my drivers. Now it's working thanks

closed 2021-02-22 by huge-11