betaflight/betaflight-configurator #2435: Filter settings partially overwritten on save #10.8.0

AndreaCCIE, 2021-03-06:
1. On the nightly configurator release, any filter setting other than the sliders gets overwritten on saving
2. On the filter settings tab, change any value (filter type, enter a manual value, disable a filter) and save. It gets rolled back. Any filter change done via CLI stays in place until a GUI change, even if made on the PID Profile Settings.
3. Provide as much information as possible about your hardware and software, including:
- it does happen on every single FC I do have where 4.3.0 is running on

AndreaCCIE, 2021-03-06:
I think this will be fixed with #2429.

spatzengr, 2021-03-07:
@AndreaCCIE , correct.  This behavior will be resolved once this PR is Merged.  I think it is ready, but needs an approval.

AndreaCCIE, 2021-03-07:
> @AndreaCCIE , correct. This behavior will be resolved once this PR is Merged. I think it is ready, but needs an approval.

Thanks Mark, is this issue present even in the CMS?

spatzengr, 2021-03-07:
I have not tested CMS, but I can think of no reason whay it would be present in CMS.  The Configurator behavior is because it was never coded to handle firmware based filter sliders yet.  My PR adds that code.

closed 2021-03-08 by AndreaCCIE