betaflight/betaflight-configurator #2438: Configurator telling me to upgrade to current version - Linux

jasonleekennedy, 2021-03-10:
I just purchased a new Mobula 6, downloaded the Linux BetaFlight version 10.7.0 and am getting the below warning when connecting:

> Warning
> The following problems with your configuration were detected:
> the version of configurator that you are using (10.7.0) is older than the firmware you are using (4.3.0).
> Using a newer version of the firmware with an outdated version of Configurator means that changing some settings will  result in a corrupted firmware configuration and a non-working craft. Furthermore, some features of the firmware will only be configurable in CLI.
> Betaflight Configurator version 10.7.0 is available for download online, please visit this page to download and install the latest version with fixes and improvements.
> Please close the configurator window before updating.
> Please fix these problems before attempting to fly your craft.

It appears that something related to the version checking is not working as expected

danybd, 2021-03-10:
I looks to be the same behaviour in android. I reported just now the same issue in #2439

haslinghuis, 2021-03-10:
If the firmware version is indeed using 4.3 which is not public released yet you indeed get this warning. The configurator has been updated to support the firmware changes, so when using an older configurator it has no knowledge about changes made in latest firmware. In short you need to install 10.8 to use 4.3, but as it's still in development it's at your own risk, but it should be very stable as we are close to go for the first release candidate. Could be some months.

jasonleekennedy, 2021-03-10:
ok, so the quad has a BF firmware unsupported by the configurator? should I downgrade it (this is what it came shipped with)?

haslinghuis, 2021-03-10:
You can use a nightly build if you are confident. Not sure why it's shipped that way.

mikeller, 2021-03-10:
@jasonleekennedy: @haslinghuis is correct - the firmware installed on your flight controller is an unstable development version - unless you want to do testing on this version, the recommendation is to use the 'Firmware Flasher' tab to install the latest released stable version of the firmware.

danybd, 2021-03-15:
@jasonleekennedy did work the solution for you? I tried the nightly 10.8.0 and the issue is fixed for me.

jasonleekennedy, 2021-03-16:
yes the nightly 10.8.0 removed the error message

haslinghuis, 2021-03-16:
as expected 😆

haslinghuis, 2021-03-16:
@jasonleekennedy as expected. Thanks for confirmation.

jasonleekennedy, 2021-03-16:
resolved.

closed 2021-03-16 by jasonleekennedy