betaflight/betaflight-configurator #2439: Android version 10.7.0 shows warning - tool not up to date

danybd, 2021-03-10:
I´ve installed the last betaflight configurator in a android device. Nevertheless I got always a warning after starting the application that my betaflight configurator is not up to date and I shoud check for new versions. As far as I know 10.7.0 is the last release and therefore I suppose this seems to be a bug.
My setup is: Android 10 & Betaflight Configurator 10.7.0

haslinghuis, 2021-03-10:
duplicate of  #2438 or the other way around :smile: 

Please look for existing issues before submitting duplicates.

mikeller, 2021-03-10:
Closing as duplicate of #2438.

closed 2021-03-10 by mikeller

danybd, 2021-03-10:
Maybe is not exatly the same described in #2438. This is the text I get in Android:

"Notice
You are using an outdated version of the Betaflight Configurator.
Using a newer version of the firmware with an outdated version of Configurator means that changing some settings will result in a corrupted firmware configuration and a non-working craft. Furthermore, some features of the firmware will only be configurable in CLI.
Betaflight Configurator version 10.7.0 is available for download online, please visit this page to download and install the latest version with fixes and improvements.
Please close the configurator window before 
 updating"

haslinghuis, 2021-03-10:
@dan557 It's the firmware which is newer than expected by configurator causing the message

haslinghuis, 2021-03-10:
FC firmware yes. Didn't see that before, maybe you have to upgrade

danybd, 2021-03-10:
But 10.7.0 is actually the last stable release. Upgrades do not even exists (apart from nightly)

haslinghuis, 2021-03-10:
You try an development APK from here: https://github.com/betaflight/betaflight-configurator-nightlies.
But as it's an development build please beware of possible issues but it should be fairly stable nonetheless.

danybd, 2021-03-10:
I've installed the last current nightly v20210303.2125 and there is no issue in the 10.8.0. If somebody tries to do the same, the old betaflight configurator version must be manually uninstalled, otherwise you wont be able to install the new one.
Thank you @haslinghuis for the support!