betaflight/betaflight-configurator #2442: Android.apk issues

pmapeso, 2021-03-14:
Android.apk preview will not connect to any of my flight controllers. The application only crashes and gives crash message.

danybd, 2021-03-15:
Could you provice more context?:
- Android version?
- Which flight controllers?
- Working in PC?
- Did you try another android device?
- Using OTG cable?
- Do another adroid apps connect with FCs?

mikeller, 2021-03-15:
Also: hardware (phone / tabled) make / model?

pmapeso, 2021-03-15:
I found that usb_hid_cdc was on on my MATEKF411 and F722-SE(matek) and that was the reason why it crashed. I turned it off via betaflight 10.7 windows 10 and betaflight on my Galaxy Note 8 worked fine. When I enable it on the phone through CLI. It kicks me out and when I try to connect I can't get back in

mikeller, 2021-03-15:
Ah yes, this is a known problem, and a duplicate of #2061 (see there for an updated version to test), and will be fixed in the next release.

closed 2021-03-15 by mikeller