betaflight/betaflight-configurator #2443: 10.8 Issue MatekF411RX

SPCORRIV, 2021-03-16:
10.8.0 on Big Sur
        **MatekF411RX** Issue when flashing board to 4.2.8 Flash stops and says board protected wait unprotecting, and just says there and does nothing but if you try to go back it says wait until current job is finished. If you unplugged  10.8.0 will not recognize board anymore.

Fix go to 10.7 flash in boot mode to recover.

Thanks

mikeller, 2021-03-16:
Just tested on MacOS BigSur, flashing a MATEKF411RX with 4.2.8 with the latest 10.8.0 configurator works fine for me:

[image]

SPCORRIV, 2021-03-16:
Maybe  Iflight issue both board were Iflight .

mikeller, 2021-03-23:
@SPCORRIV: What boards were they? Not sure the MATEKF411RX target is the correct one.

SPCORRIV, 2021-03-23:
Hey Michael

   Here's a link

https://shop.iflight-rc.com/index.php?route=product/product&path=20_27_132&product_id=1249

Thanks

haslinghuis, 2021-03-23:
@SPCORRIV yes the `MATEKF411RX` looks to be the right target according the link.

Did you enable full chip erase while flashing?
Could you also try an earlier firmware version like 4.2.6
Please press F12 and provide us with the log in the console tab when trying to flash.

mikeller, 2021-03-23:
@SPCORRIV: Ah yes, the firmware seems to be correct - I wasn't aware that iFlight had a board that uses MATEKF411RX.

But can you please supply the build version of your configurator (bottom right, after the version number)?

SPCORRIV, 2021-03-23:
Hey Michael

   I will look when I get home hope it's still there I just jumped to the
new nightly build.

Scott

closed 2021-04-30 by github-actions[bot]