betaflight/betaflight-configurator #2445: Feature Request: Load dump file into BR configurator

Sylvester411, 2021-03-18:
Good day,

I would like to request a new feature on Betaflight Configuration.  I would like to be able to load a "dump all" from someone else into BF configurator and be able to browse the GUI to see the configuration without having the flight controller attached to the serial port and the computer.  Sort of an "open file" instead of a seriel connect.

I want to help noob people having difficulties with their quad, and this gui tool would be really helpful.

Thanks.  André.

McGiverGim, 2021-03-18:
It's difficult to load a "dump" file, this has been discussed before, but in the nightly version you can find a virtual FC that can load a backup of the settings:

https://github.com/betaflight/betaflight-configurator/pull/2283

Sylvester411, 2021-03-19:
All right, thank you.  I will give this a try.

closed 2021-03-19 by Sylvester411