betaflight/betaflight-configurator #2450: beta flight in Chinese how do I get English beta flight

chiefwc, 2021-03-21:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

haslinghuis, 2021-03-21:
At the bottom of the opening screen (do not connect) you can change language.

haslinghuis, 2021-03-21:
Duplicate of #2405

closed 2021-03-22 by mikeller