betaflight/betaflight-configurator #2458: BetaFlight Configurator 10.8.0 beta

sasplayers, 2021-03-31:
### Is your feature request related to a problem? Please describe.

Firstly The new look and configuration in the configurator is excellent, the issue I had was in the CLI. I had to change 
resource motor 6 none
resource led_strip 1 A08
---ERROR---Parse error
on my "matek F405 mini" flight controller

### Describe the solution you'd like

Other than the Parce error to be fixed I think its perfect

### Describe alternatives you've considered

I did fix it via the betaflight 10.7.0 version

[image]

haslinghuis, 2021-04-02:
@Asizon   @McGiverGim any idea?

McGiverGim, 2021-04-03:
No, I can try Monday to see if I can find the bug.

closed 2021-05-17 by mikeller