betaflight/betaflight-configurator #2459: STM32 Timeout during verification (X_RACERSPI)

shyney7, 2021-03-31:
I keep getting STM32 timeouts after the firmware was flashed and the configurator switches to the verification process. 
After connecting it to the configurator everything seems to work fine. So the flashing itself seems to have worked just fine.

- Board: X-Racer F303 SPI v3.1
- Flashed in bootloader mode (boot button)
- Full chip erase
- Manual baud rate 256000
- Target: X_RACERSPI
  Version: 4.0.6
  Binary: betaflight_4.0.6_X_RACERSPI.hex
  Date: 01-09-2019 03:35
- Configurator 10.7.0 (4f646390)
- CP210x Driver Version 10.1.10.103 (08.01.2021)
- Windows 10 Build 19042.867

Although the flash process seems to have worked I think verification is also important otherwise it would not be a feature!?

shyney7, 2021-05-04:
This is still an problem and should not be closed.

haslinghuis, 2021-05-04:
That issue is almost 5 years old. As @borisbstyle said this is not firmware related as it if happens with some boards but flashing always works.

shyney7, 2021-05-05:
@haslinghuis Yes as @borisbstyle mentioned in betaflight/betaflight#455 it's configurator related that's why I opened this issue in this repo.

McGiverGim, 2021-05-05:
@shyney7 this seems some kind of "problem" between Configurator and some board. Lamentably we don't support F3 boards (like yours) anymore, and it does not happen with my F4s or F7s boards. Without some dev than can reproduce the problem I don't think we can fix it.

shyney7, 2021-05-05:
@McGiverGim @haslinghuis Is there maybe a way I can provide any usefull information like debug logs etc. to help identify the problem. I know that since the flashing itself seems to work and it's "just" the verification process its not a critical problem and I also know that you guys are already pretty busy and doing a fantastic job providing the RC Community with such a great Tool and FW's. 
But if there is a way that I can at least provide any usefull Information I would gladly do so.

haslinghuis, 2021-05-06:
A `console.log` with a nightly build could help, but still without having the issue on any F4, F7 or H7 it's hard to fix like @McGiverGim said. And if it's connecting anyhow the verification should not matter as it's working for current hardware.

closed 2021-06-13 by github-actions[bot]