betaflight/betaflight-configurator #2467: Cannot remap motors in 10.8.0

SebastianKumor, 2021-04-05:
I am trying to remap motors in the configurator i built from latest master and get this error with this command `resource motor 1 a09`

[image]

I tried latest official release 10.7.0 and it works

In the console output the pin number is omitted once your write the command and press enter.

I am running  betaflight 4.2.8

haslinghuis, 2021-04-05:
Duplicate of #2458 

Thanks for reporting, fix is ready and tested and implemented soon.

SebastianKumor, 2021-04-05:
@haslinghuis cool, I will close it then. Thanks for quick reply

closed 2021-04-05 by SebastianKumor