betaflight/betaflight-configurator #2474: Autoconnect not working (needs delay to until connect works)

redhonu, 2021-04-12:
### Is your feature request related to a problem? Please describe.

The com port is detected, before I can actually use the "connect" button, when pluging-in the FC to USB. This also means that auto connect does not work.

### Describe the solution you'd like

In the setting menu have an option to add a delay to autoconnect: like 0.5 or 1 second.

### Describe alternatives you've considered

If autoconnect failed to make a connection, automatically retry after 0.5 or 1 second.

redhonu, 2021-04-12:
Yes, I should have opned this issue for the confiurator.
Here is a [google drive](https://drive.google.com/file/d/1XfZFsnHkIEkUy8s46MO7HbBX1UILKUU2/view?usp=sharing) link to a video showng my issue.

McGiverGim, 2021-04-12:
@redhonu as suggested by @haslinghuis , please test the nightly of Betaflight Configurator that you can find here https://github.com/betaflight/betaflight-configurator-nightlies/releases it has a lot of improvements over the auto connect feature, to see if it is fixed or needs more work.

redhonu, 2021-04-12:
The nightly build is showing the exact same behavior

McGiverGim, 2021-04-12:
In the code, now it has a delay of 100ms, I suppose you say that you thing it will need a bigger delay:

https://github.com/betaflight/betaflight-configurator/blob/9ece824971ee7c537a4e3187fd718353328759e7/src/js/port_handler.js#L177-L179

redhonu, 2021-04-12:
Yes, I believe that would fix my issue.

haslinghuis, 2021-04-12:
Would a timeout option for 100, 500, 1000 and 1500 work for you so it could benefit others too?
The timeout is only used when plugging in your flight controller.

McGiverGim, 2021-04-12:
I think we can generate a version with a little more delay (250 for example) and test it. After that we can decide if make it configurable or simply change it.

haslinghuis, 2021-04-12:
Have tested with above options using options.tab, but can do a small fix first. Seems we are not collecting analytics on this. maybe in other files, didn't look further.

McGiverGim, 2021-04-12:
@redhonu you can test the fast fix here https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=2491&view=artifacts&pathAsName=false&type=publishedArtifacts

@haslinghuis if you have a version configurable push it without problem. I was saying the fast fix to simplify things.

redhonu, 2021-04-13:
I have tried the the fix, but sadly its still not working for me.

McGiverGim, 2021-04-13:
@haslinghuis if you have ready the code with "configurable" timeout update your PR and let try this.

If not simply put something big, like 1000 or 1500 to be sure that with more timeout it will work.

haslinghuis, 2021-04-13:
New code has arrived.

McGiverGim, 2021-04-13:
@redhonu can you test this? https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=2495&view=artifacts&pathAsName=false&type=publishedArtifacts

It has an option with configurable timeout.

redhonu, 2021-04-13:
I have tried it, and i have about a 10-20% success rate now at 1500ms. Would it be possible to ad an option of 5000ms timeout?
I am trying to get another PC and try it there.

haslinghuis, 2021-04-13:
Added a few extra options (2500, 5000, 10000)

redhonu, 2021-04-13:
With 2500 or anything above it always worked.

haslinghuis, 2021-04-13:
Removed 7500

redhonu, 2021-04-13:
Thank you for fixing this issue so quickly. You guys are amazing.
Should Close The issue now?

McGiverGim, 2021-04-13:
No, it will be closed automatically when we merge the fix into master branch.

PR #2475 (2021-06-26): Make connection timeout user-configurable
Fixes: #2474 
Fixes: https://github.com/betaflight/betaflight-configurator/issues/416 

Configure connection timeout in Options tab for:

- bus initialization when user plugs in flight controller with auto-connect enabled 
- save and reboot
closed 2021-06-26 by blckmn