betaflight/betaflight-configurator #2481: CLRacing F7 v2.2 - Onboard SD Card Fatal Error

Zuldan, 2021-04-30:
On the following versions I received "Fatal Error Reboot to retry error" on my brand new CLRacing F7 v2.2 FC (has not had any soldering yet, just plugged into USB).

Betaflight Configurator 3.7.0 using Betaflight v4.2.8
Betaflight Configurator 3.8.0 using Betaflight v4.3.0

[image]
.0

McGiverGim, 2021-05-03:
What I see is that your FC has a dataflash, and not a sdcard. This is the reason why it shows an sdcard error. Why it detects the option of an sdcard I can't say. I see your target defines this:
https://github.com/betaflight/unified-targets/blob/8b17caba608ecfccf55a15d391c9fb5a2a26e2ae/configs/default/CLRA-CLRACINGF7.config#L44-L45

I don't know too much about what this commands do, maybe something with more experience in the resources part can help.

Zuldan, 2021-05-03:
@McGiverGim I think you maybe have found the problem. 

Also, set sdcard_detect_inverted = ON
Might also be contributing to the problem.

I’ll do some testing and create a pull request if I find the root cause.

Zuldan, 2021-05-05:
@McGiverGim I think the link you mentioned is referencing to an older version of Betaflight. It looks like the latest version of this config file doesn’t have any references to a SD Card.

https://github.com/betaflight/betaflight/blob/master/src/main/target/CLRACINGF7/target.h

So maybe the error is still with Configurator?

McGiverGim, 2021-05-05:
Is exactly the opposite. Since "unified targets", the targets in the Betaflight repo are not used. Except a little group of them (H7 processor and little more) or to do some test while developing. Now when you install the firmware from the Configurator, it installs the "unified" hex, that is valid for all the FC/boards with the same CPU, and then customizes it your FC with the target file associated a your FC (the link of the unified-targets repo, is installed when the Configurator asks to install custom defaults for your board).

If you look at the list of assets attache to the release, you will see there is no hex for CLRacingF7 or similar.

etracer65, 2021-05-09:
There is no problem, nothing to fix, and everything is working as designed. The manufacturer has set up the target to support variants of the flight controller that have either onboard flash or a SD card slot. Since your board has onboard flash, simply select that option for the blackbox device and ignore the warning about SD card initialization.

Zuldan, 2021-05-10:
@etracer65 thank you for the explanation, that makes sense. I'll close the issue. 

Could you tell me which file I need to modify so I can compile my own hex so that the warning doesn't appear?

etracer65, 2021-05-10:
As I said, there is no problem and no changes are needed. Since your board doesn't have a SD card you can simply ignore the SD card section. It affects nothing and everything will work as expected.

However if you really can't handle the warning being displayed, then simply change this in the CLI (no need to compile a custom version):
```
set sdcard_mode = OFF
save
```

Zuldan, 2021-05-11:
@etracer65 thanks Bruce, the command worked perfectly.

closed 2021-05-11 by Zuldan