betaflight/betaflight-configurator #2484: "Failed to load extension" Error message when running from yarn.

hydra, 2021-05-02:
Fresh checkout of master and launch results in error message:

"Failed to load extension from: . Manifest file is missing or unreadable"

Steps to reproduce: `yarn start`

OS: OSX BigSur fresh install.  Also happens on my OSX Catalina laptop.

See attached screenshot of fresh install and launch.

[image]

The actual configurator binary does work, just the launch via yarn fails.

i.e. `open ./debug/betaflight-configurator/osx64/betaflight-configurator.app` works.

haslinghuis, 2021-05-02:
https://betaflightgroup.slack.com/archives/CKTRF0342/p1616191182032800

haslinghuis, 2021-06-02:
@chmelevskij have no clue about this. But if it's only on development and not on release it's not a big issue.

hydra, 2021-07-11:
@haslinghuis it's a big issue if you're trying to develop on the configurator and it won't start!

haslinghuis, 2021-07-11:
Yes I agree. But afaik after pressing OK configurator is loading. But we need to know the root cause of this message as I couldn't find one yet, but like to resolve this. We don't have a manifest file currently and this seems to be OSX specific. Just like Node 14 is working fine on Fedora 34 but not on other platforms.

hydra, 2021-07-12:
After pressing OK it does not load up.  Yes, OSX specific.

mikeller, 2021-07-13:
@hydra: Have you checked the launcher? For me it does load in OSX, but it loads minimised, so you have to click the launcher icon to get the main window to open after clicking away the error: 
[image]

chmelevskij, 2021-07-13:
Issue seems to be with [vue devtools](https://github.com/betaflight/betaflight-configurator/blob/5ed5d048317e8cfc322458b32dbe125872345425/package.json#L10)

Error message is pretty much chrome error. Tried removing it and it doesn't throw error anymore, but it doesn't open up the app as well 🤷‍♂️

chmelevskij, 2021-07-19:
App not opening up seems to be some `Big Sur` issue.

```
{ Error: Command failed: open debug/betaflight-configurator/osx64/betaflight-configurator.app
The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10827 "kLSNoExecutableErr: The executable is missing" UserInfo={_LSLine=3691, _LSFunction=_LSOpenStuffCallLocal}

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:198:13)
    at ChildProcess.EventEmitter.emit (domain.js:466:23)
    at maybeClose (internal/child_process.js:982:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:198:13)
    at Socket.EventEmitter.emit (domain.js:466:23)
    at Pipe._handle.close (net.js:607:12)
  killed: false,
  code: 1,
  signal: null,
  cmd:
   'open debug/betaflight-configurator/osx64/betaflight-configurator.app' }
   ```
   
   Also, it seems to work even when doing `open debug/betaflight-configurator/osx64/betaflight-configurator.app` right after failure.

haslinghuis, 2021-07-19:
Please try: https://disable-gatekeeper.github.io/

hydra, 2021-07-21:
Just did a fresh install on another OSX machine (Macbook Pro 2015) and ran into the same issue today.

@haslinghuis 

> xattr -dr com.apple.quarantine

doesn't help here.

@chmelevskij I was encountering the problem when I had OSX Catalina installed.  My old laptop died so I no longer have a working Catalina install.

haslinghuis, 2021-07-22:
Big Sur 11.5 is out now. Maybe try disable Gatekeeper system-wide and reboot to confirm this is not the issue.

chmelevskij, 2021-07-23:
As I've mentioned, these are 2 separate issues.

`failed to load extension` happens because of vue devtools extension, removing the env variable which loads it gets rid of that error.

Disabling gatekeeper doesn't really make sense. Just using `open` works fine, but for some reason when it is run from gulp task it gets that error.  So it must be some sort of issue with `node` running `open` for us.

mikeller, 2021-07-23:
@chmelevskij: Thank you for the detailed analysis. I think, all things considered, and seeing that everything effectively works, and all we have is an unwanted error message when running in debug mode, the right way out of this is to document the error message and focus on things that provide more value to the end user.

closed 2021-08-31 by github-actions[bot]