betaflight/betaflight-configurator #2488: Betaflight Configurator v10.8.0 (Nightly) PID Sliders need polishing

Zuldan, 2021-05-11:
Currently in Betaflight Configurator v10.8.0 Nightly (v20210413.2497), if you have the PID sliders enabled and manually change the PID values then click on “Save”. The current slider positions and their associated values will override the PID values you manually entered in. I’m not sure this is the expected behaviour. I think the sliders should automatically be disabled when you manually enter in PID values OR make the PID values read-only when the sliders are enabled. Unfortunately my Javascript knowledge is extremely limited so I can’t create a PR myself.

haslinghuis, 2021-05-11:
[image]

Have to set RPY to OFF to be shut off the sliders keeping there last value to be able to go in expert mode.
Agree this could be more transparant like legacy mode.

haslinghuis, 2021-05-11:
[image]

Have to set `RPY` to OFF to be shut off the sliders keeping there last value to be able to go in expert mode. This is the way how it works right now (tune with sliders first, set them OFF and be able to change manual after this)  but agree this could be made more transparent like legacy mode.

closed 2021-06-19 by github-actions[bot]