betaflight/betaflight-configurator #2491: Android App Issue - 'Invalid File Extension'

SupaflyFPV, 2021-05-17:
I'm having an issue trying to load or restore files in the CLI section of the Andoird App. On the desktop App I can load .txt files with CLI commands no problem, but the same file on the android app says 'invalid file extension'

I can get to give the same error even when saving a eg 'diff' file from the android app, then trying to reload it. It says 'invalid file extension' for it's own file...

Im using an Aikon F4 FC with a Google Pixel 4a. All other configurator aspects and cables working fine...

Let me know if there is anything I need to try...thanks!

haslinghuis, 2021-05-17:
Assuming you are using the 10.7 Android release. Can you please try `file` functionality on other tabs? Like to know if any file function is working on Android.

Besides Android is broken in master probably due to an incorrect MIME setting. Nightly build `1030.1206` is the last known working version for 10.8 dev. Already was working on replacing the chrome API with NWjs but this doesn't work with Android. So I am betting on File System Access API (with the inclusion of `startin` parameter to remember last working directory) but it's not that easy because the surrounding code depends on the Chrome API File Entry which would be replaced with a File Handler needing changes in code relying on the old Entry.

File functions:
1. cli
2. blackbox
3. firmware flasher
4. video transmitter
5. tethered logging
6. backup and restore
7. font manager

haslinghuis, 2021-05-17:
Ping @WalcoFPV

closed 2021-06-30 by github-actions[bot]