betaflight/betaflight-configurator #2499: PID Slider: PD balance changed ?

muhkuh0815, 2021-05-30:
Why did the PD Balance Slider changed in the nightlies ?

before with 10.7.0 and some older 10.8 nightlies   P was fixed and the PD balance slider adjusted D, now D stays the same and P changes ...

just got into brian white's 2-step basement tuning from the pidtoolbox youtube channel (https://urp.mak-it.at/basement-tuning/ ), worked pretty great until i changed to the newer nightlies where the pd balance slider is changed :( .

haslinghuis, 2021-06-01:
We had some discussion already with @spatzengr. @IllusionFpv should know and suggested some changes, but it will take some time to get ready.

spatzengr, 2021-06-01:
This is by design to get back to the correct behavior of 4.1 before it was messed up in 4.2.  D should be held and want your raise PD Balance slider, it should raise P and old D constant.

haslinghuis, 2021-06-15:
Also see #2488

closed 2021-06-15 by haslinghuis