betaflight/betaflight-configurator #2500: Wiki needed how to install configurator on Linux

and-sh, 2021-05-31:
I'm new to Linux, but I really like this OS. 
It took me a few days to get the configurator to work.
For example:
install JRE
sudo usermod -a -G dialout $user
remove modemmanager
remove putty
I suspect that some of the actions were redundant.

Is it possible to collect a list of problematic and necessary modules and actions in one place?

haslinghuis, 2021-05-31:
See https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md or https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Fedora.md

and-sh, 2021-06-01:
Thank you. It helps a lot.
By the way, somewhere there is a description of how to make my target.
For my some modules are missing in unified target, some are unnecessary.

closed 2021-06-15 by haslinghuis