betaflight/betaflight-configurator #2509: Feature request: enable SA to work in parallel with MyVTX

MadAEng, 2021-06-12:
If what Trappy told me is correct, a BF FC continuously sends the band/channel/power settings to the VTX, so hooking SA up to both a crossfire RX and an FC is pretty much pointless, cuz the last command will always come from the FC. But even if that did work, there are still a few things that I would probably want that would not be currently available.

The essence of the functionality I'm looking to add is to set up the band/channel/power before powering up my quad so I don't stop on someone else's video when I'm flying with others.

On it's own, MyVTX is the way to go for that, but in that case I'm missing out on some great features in BF, hence this request. Not sure if all of this is even possible hardware-wise, but here's what I would love to be able to have.
- Have band/channel/power on the OSD, no matter which device (RX or FC)  triggers the settings.
- Output power on a switch is something that I don't think I can do with just MyVTX, but it does work with BF, definitely want that one.
- Parallel to, instead of, or in combination with the previous point: I love the 'low power when unarmed'-feature. Depending on how it is implemented, any way to prevent frying a VTX is pretty much a must.
- Have all of that working with a pre-power up setting of band/channel/output power.

Would any of this be at all possible? Either by hooking them up to the VTX in parallel or perhaps by using a UART to send the RX SA to the FC?

McGiverGim, 2021-06-17:
I don't know if what you want can be done, but this feature request must go in the "firmware" repo. This is the "Configurator" repo.

The firmware repo is here: https://github.com/betaflight/betaflight

Search first there, maybe someone has suggested that before.

JekleFPV, 2021-06-17:
Thnx, I'll do that!

(but it would probably affect the configurator too I would suspect ;) )

closed 2021-07-25 by github-actions[bot]