betaflight/betaflight-configurator #2510: Unable to flash firmware from local file

MartinWa, 2021-06-13:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
I wanted to downgrade to Betaflight 3.5.7 and it is not available in the dropdown. Instead I downloaded the firmware from GitHub. I loaded it from local file but the "Flash Firmware" button did nothing. No logs showed errors.

I downloaded Betaflight Configurator 10.5.1 and it worked. (10.6.0 and 10.7.0 did not)

Another person created this Reddit post on what seems to be the same problem:
https://www.reddit.com/r/fpv/comments/mkvzqw/cant_flash_local_firmware_file_in_betaflight/

haslinghuis, 2021-06-14:
Not able to reproduce with 10.7.0 or 10.8.0 on Windows (needs drivers) or Linux (needs rules and add yourself to dialout and plugdev group and remove modemmanager). Look at our documentation for more information.
Ubuntu example:
```
    $ sudo usermod -a -G dialout $USER
    $ sudo usermod -a -G plugdev $USER
    $ sudo apt-get remove modemmanager
    $ (echo '# DFU (Internal bootloader for STM32 MCUs)'
     echo 'ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"') | sudo tee /etc/udev/rules.d/45-stdfu-permissions.rules > /dev/null
```
On Windows you will need other drivers, like Immersion RC driver fix or Zadeg.

closed 2021-06-14 by haslinghuis

MartinWa, 2021-06-15:
@haslinghuis 
I am sorry but the original post had been chopped off by mistake. Also it did not contain enough information. So:

On Windows 10 using the betaflight-configurator-installer_10.7.0_win32.exe installer I have an issue with flashing firmware from local file. I can flash without problem firmware from the dropdown list (for instance the latest) but for older firmware that is not in the list (in my case Betaflight 3.5.7 that I wanted to test the original settings that came with my drone) there is no dropdown.

The button to flash does not do anything when I load firmware from local file. The problem exists in 10.6.0 but not in 10.5.1.

Perhaps this is intentional (that 10.6.0 no longer supports flashing 3.5.x) but then I was hoping for an alert for this fact instead of no action at all (or logs).

haslinghuis, 2021-06-15:
Did not find anything which would block flashing an offline hex file.
Maybe you can try the maintenance 3.2 release from the dropboxes.
We stopped supporting releases before 4.0 a year ago, but did not make it impossible to use.

McGiverGim, 2021-06-15:
Exactly, the flashing must work for all versions. Maybe you can record a video with what you do to see if we can reproduce it?

BurnoutAg7, 2021-09-14:
I encountered the same issue, but found out that I didn't leave the FC board sector blank.

(Just a small tip for anyone who is kind of dumb one day like me.)