betaflight/betaflight-configurator #2515: Problem with M1 macbook crashing

michael-blue2, 2021-06-20:
When I open the configurator, and I have the quad connected it crashes (photo bellow) and it doesnt even open. When I dont have anything connected to the MacBook then the configurator open up normally. I can then connect the quad. connect to the settings but for a few seconds (max 20s) and then the app it crashes again and i get this error.

I am using a MacBook pro m1 8gb with version 11.4
betaflight configurator is 10.7

[image]

haslinghuis, 2021-06-20:
Duplicate of #2274

closed 2021-06-20 by haslinghuis

haslinghuis, 2021-06-22:
@michael-blue2 did you check the nightlies build: https://github.com/betaflight/betaflight-configurator-nightlies/releases