betaflight/betaflight-configurator #2520: Feature Request: User configurable presets and Custom tab

FPVian, 2021-06-28:
Firstly, I believe there should be a "Custom" tab where users can add parameters that could previously only be adjusted through the CLI. This way, if you adjust something like vbat_sag_compensation or thrust_linear on all of your builds, this can easily be added into the configurator on a case-by-case basis.

Second, I believe there should be user-defined presets, so that we can save the configuration changes we make to all our quads and apply them using a dropdown menu (possibly in the configuration tab). For instance, I could save all the changes I've made on a specific flight controller and call it my "5 in. freestyle" preset. As an added bonus, allow configuration of which settings are saved or not. Here's a list of settings that are similar across many of my quads, that need to be changed on every new flight controller:

feature
beeper
beacon
rxfail
profile
rateprofile 
aux
adjrange

set osd_XYZ_pos
set osd_warn_XYZ
set osd_stat_XYZ
set vtx_XYZ
set vcd_video_system
set yaw_motors_reversed
set serialrx_provider
set dshot_bidir
set motor_pwm_protocol
set small_angle
set vbat_warning_cell_voltage
set debug_mode
set align_board_XYZ
set rssi_channel
set gyro_lowpass2_hz 
set dyn_notch_XYZ
set dyn_lpf_gyro_XYZ
set min_check
set max_check
set deadband
set yaw_deadband
set report_cell_voltage
set vbat_sag_compensation
set iterm_windup
set pidsum_limit
set pidsum_limit_yaw
set dyn_lpf_dterm_curve_expo
set ff_interpolate_sp
set ff_boost
set gps_rescue_min_dth
set thrust_linear

Yes, I do usually just copy paste everthing in the CLI, but I think it would increase quality of life for newer users to be less dependent on the CLI for day-to-day tasks.

haslinghuis, 2021-06-28:
Duplicate of #1811

Please also see: 
https://github.com/betaflight/betaflight-configurator/issues/1139
https://github.com/betaflight/betaflight-configurator/issues/2167
https://github.com/betaflight/betaflight-configurator/pull/2489

closed 2021-06-28 by haslinghuis

FPVian, 2021-07-01:
I would like to point out that the idea of having a "custom" tab where users could add variables that are only available in the CLI is an original idea, or at least it wasn't covered in any of the links.

Also this feature request is slightly different because the presets would be stored locally, not online, so they could include receiver, configuration, and modes changes without worrying about somebody else accidentally using it as a "pid tune." I think #1811 is great, but it should include an offline option for users who build multiple quads that are almost identical to each other.

Hope that clarifies things. Thank you.

haslinghuis, 2021-07-01:
@limonspb Any thoughts?