betaflight/betaflight-configurator #2526: New firmware detection code doesn't work for new targets.

hydra, 2021-07-01:
1. Describe your problem;

Incorrect error message as follows:

[image]

2. Include ways to reproduce the problem;

Flash an existing target for which there is a release
Create a new firmware target (no firmware published for it yet)
Observe that the 'Select your board' dropdown is BLANK.
Observe that the 'Firmware version' is NOT BLANK and should be.
Click 'Load firmware local'
Load up the .hex for the new firmware target.
Click Flash Firmware.
Observe error message shown as above

I noticed this when flashing the SPRacingH7RF local firmware image after flashing an SPRacingF7DUAL.

3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
Windows x64

- the version of all the software used;
configurator built from latest `master`.
- how things are connected / wired up.
n/a

hydra, 2021-07-01:
@haslinghuis this is related to #2485

Asizon, 2021-07-01:
@hydra In this case, for development you need to add config file using cli, like before if the target config is not uploaded to github.

Asizon, 2021-07-01:
Also please test with https://github.com/betaflight/betaflight-configurator/pull/2524

hydra, 2021-07-01:
> 
> 
> @hydra In this case, for development you need to add config file using cli, like before if the target config is not uploaded to github.

The error message says I've *SELECTED* something that I have not, indicating that the implementation doesn't handle this case.

I don't understand the "like before" part of your reply.  I've never had to make any local changes before for every new target that I've ever made, and I've made A LOT!  I think it's unreasonable to force a target developer to work around errors in the configurator that are incorrect and shouldn't appear anyway.

Asizon, 2021-07-01:
@hydra i hope https://github.com/betaflight/betaflight-configurator/pull/2524 fix your issue :)

PR #2524 (2021-08-01): Detect board in firmware_flasher patch
Patch for firmware flashing detect board feature introduced in #2485

Fixes: #2526
Fixes: #2498 

After much testing this patch provides the following changes (thanks for all feedback):

- Removes board verification on tab load and instead let the user check with a button reducing the connection load to avoid connection issues.
- Slapped a timer on the button to prevent spamming as requested by @Asizon
- Button is much smaller now and next to board selection as requested by @McGivergim
- Only present a dialog (when using MSP capabilities) in `msp32.js` wh [... cut]
closed 2021-08-01 by blckmn