betaflight/betaflight-configurator #2528: [10.8.0] DANGEROUS motor going to 100% throttle instead of stop

nablabla, 2021-07-01:
until version 10.7. the home button made motors full throttle and the end button made them stop. In 10.8. it is the other way round!!!
Talking about the Motors tab, you check the "i am not an idiot"-switch then you can click on a slider to change throttle-% or if you use page up and down you can go up and down by steps of 10% and home and end for 100%. But in 10.8 the function is reversed. I was always skeptical of the home button anyways, but the end button was handy way to stop what ever stupidity I was doing. Now I just got shocked. it might be dangerous. I suggest to get rid of the 100% button and reverse back so that End does the stopping.

PS: Of course I took the props like a fine citizen and I am totally fine :-) 
(actually i didn't, instead use a lab bench pwr supply that stopped the madness, and finally i have proven that it is sufficient to set it to 1A current limit and be a Lazy Lou)

haslinghuis, 2021-07-01:
It must be changed recently as an older 10.8 build from may this year it's indeed the other way.
But imho `home` should be 1000 and `end` / `EOF` 2000.

nablabla, 2021-07-01:
independant of what you or I think it is a bad idea to switch it, right?
I think Home shoud be 2000 and End shoud be 1000 because of how the keys are arranged on the 6-keys pad on the keyboard, pageup makes it go faster and next to it is home, page down makes it go slower and next to it is End. If you do it like you  state it should be iyho it would make some kind of cross.

haslinghuis, 2021-07-02:
We didn't switch. Checked the code and not sure which caused this. Some API or some package update most have changed this behavior.

nablabla, 2021-07-03:
hmm, i am so used to doing it like that, it just happened to me again. Noone got hurt but my whoop scraped over my desk

haslinghuis, 2021-07-03:
`Home` and `End` are dangerous and to me current behavior seems more right. It would be better to disable the buttons.
Anyway preference aside it was indeed a package update found here: https://github.com/betaflight/betaflight-configurator/pull/2507

haslinghuis, 2021-07-04:
Already attempted to disable the keys but doesn't work for the individual sliders.
```
        const preventHomeEnd = (e) => {
            if (e.key === 'Home' || e.key === 'End') {
                e.preventDefault();
            }
        };
        document.querySelector('div.sliders input:not(.master)').addEventListener('keydown', e => preventHomeEnd(e));
        document.querySelector('div.sliders input.master').addEventListener('keydown', e => preventHomeEnd(e));
```

haslinghuis, 2021-07-06:
Checked with NW.js `0.54.1` also. Same behavior. We should check with a later version of Node but other updates are needed there.

haslinghuis, 2021-07-06:
This code disabled the Home and End button in motors tab:
```
        const preventHomeEnd = (e) => {
            if (e.key === 'Home' || e.key === 'End') {
                e.preventDefault();
            }
        };

        document.querySelectorAll('div.sliders input').forEach(e => e.addEventListener('keydown', e => preventHomeEnd(e)));
```

haslinghuis, 2021-08-01:
Closing this as new behavior seems more intuitive and we should adapt.

closed 2021-08-01 by haslinghuis