betaflight/betaflight-configurator #2530: Sliders disabled while only moving/changing sliders?

audiogek, 2021-07-02:
At following slider setting(see pic), sliders are disabled when saved, and BF configurator thinks I set the pid numbers manually.
I only moved the sliders and did not change any number manually.

[image]

haslinghuis, 2021-07-02:
Cannot reproduce with 4.2.9 firmare on 10.7 or 10.8. I have fixed this behavior on 10.8 but have not worked on 10.7, only difference I notice in the screenshot is Derivative is separated as Dmax and Dmin.

[image]

[image]

audiogek, 2021-07-02:
> Cannot reproduce with 4.2.9 firmare on 10.7 or 10.8. I have fixed this behavior on 10.8 but have not worked on 10.7, only difference I notice in the screenshot is Derivative is separated as Dmax and Dmin.
>

And when you uncheck D-min?

haslinghuis, 2021-07-02:
Yes with 4.2.9 on 10.7 / 10.8 or 4.3 on 10.8 all suffer from Dmin being disabled.

haslinghuis, 2021-07-02:
@IllusionFpv as calculation in 4.3 is done in firmware now, does it take dmin setting in account?

audiogek, 2021-07-03:
> Yes with 4.2.9 on 10.7 / 10.8 or 4.3 on 10.8 all suffer from Dmin being disabled 😞

I was asking: Can you reproduce the bug I mentioned with d-min disabled?

Or do you mean you already solved this for 4.3?

haslinghuis, 2021-07-03:
No. I meant to say I can reproduce the problem. Firmware 4.3 introduces a new set of sliders which is still being worked on. Raised this issue with the team.

audiogek, 2021-07-03:
So this is basically soved. Nice, thanks!

Asizon, 2021-07-04:
Fixed in https://github.com/betaflight/betaflight/pull/9119

closed 2021-07-04 by Asizon