betaflight/betaflight-configurator #2533: "Unrecoverable Failure of serial connection, disconnecting..." Error when attempting to flash UARTs on F722 board

K0SSL, 2021-07-09:
Hi, i cant fix this error code ive been getting every time i try to flash my F7 board. I have reinstalled all drivers, used the impulseRC driver fixer, reinstalled betaflight, used different cables and so on, but nothing seems to fix it. I tried to enable another UART port for a GPS sensor on my Mamba F722 MK3 Flight controller, but everytime i try to flash and reboot i get the error message. 
Weird thing is, when i just enable the UART port (and dont select GPS as sensor output) it still gives the error message, but the ports stay enabled when connect it again. If i activate the ports AND select additional things, the configuration is not saved. Is there a fix for this? I feel like ive tried everything at this point

haslinghuis, 2021-07-09:
What target are you trying to flash?

Probably you need to import and save https://cdn.shopify.com/s/files/1/0027/2708/4144/files/MAMBA_F722_I2C_20201022.config?v=1617711515.
Please download https://github.com/betaflight/betaflight/releases/download/4.2.9/betaflight_4.2.9_STM32F7X2.hex
First select the config file you can download from the 1st link and then load the hex file you downloaded in the 2nd link for flashing your MK3.

K0SSL, 2021-07-11:
> What target are you trying to flash?
>
> Probably you need to import and save https://cdn.shopify.com/s/files/1/0027/2708/4144/files/MAMBA_F722_I2C_20201022.config?v=1617711515.
> Please download [...]

This worked, thanks :-)

closed 2021-07-11 by K0SSL