betaflight/betaflight-configurator #2538: can't connect quad through serial ports

channing1980, 2021-07-19:
So I just switched to a new MacBook Pro m1 and having connection problem with my quad. After installing CP210VCP driver, I didnt see a new serial port in the drop down menu of betaflight. Im sure I'm using good data cables. Ive tried unistalling and reinstalling. and restarting the computer. Does anyone have any ideas on how I can get my qaud to connect?

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

haslinghuis, 2021-07-19:
Please try:

- power down Mac
- hold Control Option P and R while clicking Power button and hold until you hear the 2nd chime.

haslinghuis, 2021-08-01:
Closing now as this seems a Mac problem - above steps should solve this.

closed 2021-08-01 by haslinghuis