betaflight/betaflight-configurator #2545: can't manually change PIDFs

mituritsyn, 2021-07-26:
impossible to manually change PIDF coefficients in latest configurator;
Steps to reproduce:
- Manuaдly change PIDF values, hit save button
expected behavior:
-new PIDF values are saved, sliders are hided
actual behavior:
-PIDF values changes are discarded

haslinghuis, 2021-07-26:
Duplicate of #2530

haslinghuis, 2021-07-26:
To be able to adjust PIDF manually you have to set `RPY` to `OFF`.

Design now is to use (firmware dictated) sliders (settings) first. Then put them OFF and adjust manually further if needed.

[image]

haslinghuis, 2021-07-27:
We are considering to automatically disable the sliders upon changing settings manual.

closed 2021-09-30 by mikeller