betaflight/betaflight-configurator #2551: disable autoscroll in firmware flasher tab

mituritsyn, 2021-08-04:
There is annoing autoscroll to the "Release info" on this tab. On first entering this tab it will autoscroll, then one need to scroll up to select firmware:
- check "unstable releases" - you'll get auto scroll
- scroll up check "enable expert mode" - you'll get auto scroll
- scroll up change release to release candidate - and again...
I thing this is due to autodetect function but I'd prefer the header part stay constantly on top to see what am I doing on this tab.

haslinghuis, 2021-08-04:
Let's wait for other opinions on this.

haslinghuis, 2021-10-29:
Yes - not happy with the last commit after request to make it conditional either.

haslinghuis, 2024-05-06:
@mituritsyn think we can close this.

PR #2554 (2021-09-26): Remove scrolling conditional and allow auto-detect board when no board selected
Fixes: #2551
Fixes: #2609
closed 2024-05-06 by haslinghuis