betaflight/betaflight-configurator #2555: Android 10.8.0 problem

Myllerman, 2021-08-05:
I have been using the 10.7.0 android apk on my Samsung S7 without problem.
When i try any of the nightly android apk:s they are starting up with the Betaflight logo screen as usual but then there is a loading circle that goes on forever. Nothing more happens. My phone has android 8.0 and that is the latest for this oldish phone.
Is it a problem with my old android or something else?

McGiverGim, 2021-08-05:
I've tested the latest nightly and at least in my phone (Google Pixel 4) it works. Without some developer reproducing the error and seeing the logs, I don't know what can be happening.

Myllerman, 2021-08-05:
Thanks for testing it. Atleast now i know it is working for someone else. I guess i can wait and see if newer builds might fix it. I dont know how to look at logs but i might read into it.

butterSnake, 2021-08-10:
I’m having the same issue using android 9 on a cheap tab

mikeller, 2021-08-10:
@Myllerman, @butterSnake: It's really hard to publish software that supports both current operating systems and at the same time versions that had their end-of-life 3 respectively 4 years ago - the problem is that even the libraries that we are using are dropping support for these old versions in their latest versions.

Myllerman, 2021-08-10:
> @Myllerman, @butterSnake: It's really hard to publish software that supports both current operating systems and at the same time versions that had their end-of-life 3 respectively 4 years ago - the [...]

I see, yeah its unfortunate. My newer phone is an Iphone so its not good for these sorts of applications.

mikeller, 2021-08-10:
@Myllerman: Yeah, Apple devices are a hard case for these things - iOS refuses to talk to devices that haven't been blessed by Apple.

trnnwnnhe, 2021-09-13:
I have same problem.  Configurator 10.8  not work. Redmi note 4x  android 7.0 . No conectioon to FC. Configurator 10.7  works perfect.