betaflight/betaflight-configurator #2566: Move "Motors" tab above "PID Tuning"

mituritsyn, 2021-08-16:
As we moved some options to different tabs, could we move "Motors" tab above "PID Tuning"? The point is that we need activate Esc/Motor protocol and BDshot options on Motors tab to be able fully configure PID/Filters and Modes tabs. 
So now we have following order while tuning: setup>ports>conf->skip 4 tabs to the motors to set protocol ->PID or anything else while it could correspond to tabs order from top to bottom: setup>ports>conf->power>motors->pid> etc

haslinghuis, 2021-08-16:
[image]

haslinghuis, 2021-08-16:
[image]

Other suggestions:

- Fail-Safe should go after Receiver.
- Servos perhaps after Motors

ikherty, 2021-09-17:
This suggestion will be most useful for targets that do not have dshot pre enabled

haslinghuis, 2021-11-19:
@ctzsnooze ?

closed 2021-12-26 by github-actions[bot]