betaflight/betaflight-configurator #2594: Mac - cannot flash local firmware file without closing and re-opening Configurator

ctzsnooze, 2021-09-03:
For some time now, it's not possible to flash a local firmware file to the flight controller, after entering DFU mode in CLI, then  quitting and then re-opening Configurator and then flashing

So this sequence fails:
- connect to flight controller
- got to CLI, type BL to enter DFU, save
- re-connect
- go to Firmware Flashing tab, confirm DFU mode at serial connection dropdown
- Load local firmware, click Flash
-> nothing happens.

But if we do this, it works:
- connect to flight controller
- got to CLI, type BL to enter DFU, save
- quit Configurator
- re-open Configurator
- go to Firmware Flashing tab, confirm DFU mode at serial connection dropdown
- Load local firmware, click Flash
-> flashing starts

Also there seems to be no way to enter DFU mode other than via the CLI.  If it is supposed to happen automatically by pressing one of the flashing related buttons, it doesn't

ctzsnooze, 2021-09-06:
Update...
Discovered something fascinating!
If I hold a boot button down while plugging an FC into the USB port, and release really quickly, everything works great!
If I hold the boot button more than half a second after plugging in, I have to do the quit/reboot of Configurator thing...

mikeller, 2021-09-06:
@ctzsnooze: There is also a button in the Setup tab that allows you to put the flight controller into DFU mode:

[image]

Not sure if this will be working better than `bl` in CLI on Mac.

Also, I can confirm that this problem does NOT exist on linux - setting the flight controller into DFU mode through CLI and then flashing without a configurator restart is working there.