betaflight/betaflight-configurator #2598: "Reset all profiles" doesn't work on Rateprofile settings

mituritsyn, 2021-09-04:
To reproduce:

- come to pid tuning->rateprofile settings
- change rates settings
- save
- hit "Reset all profiles" button

Expected:
- rates are reseted to default

Actual:
- rates still changed

reproduces in 10.7 and master branches

mituritsyn, 2021-10-05:
up

mikeller, 2021-10-05:
@mituritsyn: The problem here is actually misleading labeling on the button - what the button does is to reset the *PID* profiles - the rate profiles are not affected by this.

mituritsyn, 2021-10-05:
thanks for the clarification. I remember for sure that in previous versions the reset worked for rates, at least untill save is pressed, which allowed experimenting with values and easily rolling back changes. but with new label it's ok for me.

closed 2021-10-09 by blckmn