betaflight/betaflight-configurator #2605: Filter Slider not available/visible - on MacOS - NightlyBuild v20210909.3809

mrRobot62, 2021-09-14:
Environment: MacOS, 10.15.7 (19H1323) (Catalina)
Betaflight NightlyBuild : 10.8.0 - v20210909.3809
BF-FW : 4.3.0

[image]

If I go to the filter tab and click on "Enable Slider" nothing happens. No sliders come up.
Tried to do this with a nightly build from 01.09.2021 same problem on this build.
Is enabling slides depending on other things? (BTW: Expert mode is active)

Bug?

mrRobot62, 2021-09-14:
Update:
[image]
Same for the DTermSlider - value is 0

It seams there is a rule, that set both variables to zero and this is the reason why slides are not visible.
Unfortunately I don't find the "rule"

haslinghuis, 2021-09-14:
For me 4.3 master on 10.8 master with Matek F411 does work with defaults?

Thanks for the analysis, we are working on the code and new defaults as we speak.

See https://github.com/betaflight/betaflight/pull/10919 and https://github.com/betaflight/betaflight-configurator/pull/2572

mrRobot62, 2021-09-17:
Hi, i checked my firmware and I used this one for my F405
Betaflight: betaflight_4.3.0_STM32F405_5cd2ed35c.hex
Betaflight-Configurator: betaflight-configurator_10.8.0-debug-4ff6e209_macOS.dmg

Hope I understood you right, that these nightly versions are correct for testing

BTW: in [BF#10919](https://github.com/betaflight/betaflight/pull/10919) you wrote:
*Firmware for testing. Please load the config file for your board first:*
Should I do something before I flash the firmware? uups

haslinghuis, 2021-09-17:
@mrRobot62 as both PRs are in final phase. 

Yes before flashing the custom firmware from the firmware PR you have lo load your config file which you can get here: https://github.com/betaflight/unified-targets/tree/master/configs/default (use raw format).

The configurator to test this firmware is at:
https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=3902&view=artifacts&pathAsName=false&type=publishedArtifacts

mrRobot62, 2021-09-17:
ok thanks ! Will check this and test again - Earliest starting next week, has now time over the weekend.

mrRobot62, 2021-09-18:
Update. Install lates configurator as you suggested.
FW on FC F405 is night build as you suggested
Config was installed
[image]

[image]

**So, seems you fixed this bug.**
Will do some more deep dive tests on mac environment. Unfortunately I do not have other F4xxx FC as my F405er so not sure if we should close this issue.

haslinghuis, 2021-09-19:
Yes range for DTerm should be 80-120. Requested change in firmware.

mrRobot62, 2021-09-19:
New Test sequence
1) reflash FC and store config
2) check simplified commands after flashing
`simplified_dterm_filter_multiplier = 100
simplified_gyro_filter_multiplier = 100
`
3) go to filter tab
[image]
4) click on slider tab (nothing happens, no sliders, no values were changed)
5) go to CLI and change value for e.g. simplified_dterm_filter_multiplier = 255
6) go back to filter tab and check values - **no changes are visible** but they have to be
Values are exact the same as above screenshot
7) click on slider tab - no sliders were opened but values for dterm were changed
[image]

mrRobot62, 2021-09-19:
# Update 19-09
BF-FW : betaflight_4.3.0_STM32F405_55a67977f.hex
BF-Config: latest nightly build from Azure pipeline

[image]

## Test 1 : Expert-Mode OFF - PID-Filter Sliders visible
[image]
- Test ok

## Test 2: Moving sliders left/right - check calculated values
[image]

[image]

- Problem with calculation

## Test 3: same as Test 2 with DTerm sliders
- no Problem with this slider, even how fast I move left/right everything ok

# Conclusion
see #2606  - Conclusion it seems it's timing problem with fast and slow moving sliders and not a real math calculation problem. I assume it depends how many fields will be updated and recalculated during for an appropriate slider. As more fields effected as higher probability that a calculation error occured

# Good news:
the base error for this issues seems to be fixed on F405 FC !

haslinghuis, 2021-09-19:
Continued on #2608

closed 2021-09-19 by haslinghuis