betaflight/betaflight-configurator #2606: Filter Slider not visible on Kakute F4 (see #2605)

mrRobot62, 2021-09-15:
Same problem as described in #2605 with an other FC 

[image]

Filter-Sliders are not visible. Maybe same problem with default values for this FC

haslinghuis, 2021-09-15:
#duplicate of 2605

haslinghuis, 2021-09-15:
Duplicate of #2605

mrRobot62, 2021-09-15:
Hi @haslinghuis I assume, it‘s an general problem with F405 FC and default values?
Right?

haslinghuis, 2021-09-15:
Until now yes it seems F405 specific, also my Matek F722 works
As we are developing on firmware and configurator at the same time there is no point to fix this but it's useful to check why it's failing.

Maybe we see something in the settings.
Please show a 
```
diff
``` 
and a
```
resource show all
```

mrRobot62, 2021-09-16:
ah ok, so if you need some testing assistance, I can help if needed. In this case it should be helpful define a test scenario which should be testest. Makes life easier as „let‘s test everything“ ;.-)

mrRobot62, 2021-09-16:
Made a diff on my Kakute F4 (STM32_F405) and see following errors:
`set simplified_gyro_filter_multiplier = 0
###ERROR IN diff: CORRUPTED CONFIG: simplified_gyro_filter_multiplier = 0###
set simplified_pitch_pi_gain = 1
###ERROR IN diff: CORRUPTED CONFIG: simplified_pitch_pi_gain = 1###
set simplified_dterm_filter_multiplier = 1
###ERROR IN diff: CORRUPTED CONFIG: simplified_dterm_filter_multiplier = 1###

`

Tests:
`set simplified_gyro_filter_multiplier = 1` => result in  `###ERROR IN set: INVALID VALUE### Allowed range: 50 - 255 `
did the same with all other RED flagged parameters, set them to the lowest value.
Check than filter tab - with result: unfortunately no good result - Slider-Tab not visible
Than I checked again:
all my settings were ignored although I did SAVE the values.

mrRobot62, 2021-09-16:
#Update set simplifiedxxxxxx parameter
I retested it again and found following behavior

1. it is possible to save a change on simplified_gyro_filter_multiplier and simplified_dterm_filter_multiplier
2. a save in CLI and restart and a get on above Paraments shows - values were saved correctly
3. close CLI and change to PID / Filter - no Filter slider visible
4. click on Slider Button -> nothing happens
5. go back to CLI and check above parameter again
6. both parameter values were rest to wrong values 
`simplified_dterm_filter_multiplier = 1
###ERROR IN get: CORRUPTED CONFIG: simplified_dterm_filter_multiplier = 1###
simplified_gyro_filter_multiplier = 0
###ERROR IN get: CORRUPTED CONFIG: simplified_gyro_filter_multiplier = 0###
`
It seem at the moment I call PID-mask values were reseted and stored in eprom.

haslinghuis, 2021-09-16:
Multiplier should indeed be 100.

mrRobot62, 2021-09-16:
found this inside code:
`TuningSliders.resetGyroFilterSlider = function() {
    if (semver.gte(FC.CONFIG.apiVersion, API_VERSION_1_44)) {
        if (!this.cachedGyroSliderValues) {
            this.sliderGyroFilterMultiplier = 1;
        }
        this.sliderGyroFilter = 1;
        FC.TUNING_SLIDERS.slider_gyro_filter = 1;
        this.initGyroFilterSliderPosition();
    } else {
        this.sliderGyroFilterMultiplier = 1;
        $('#sliderGyroFilterMultiplier').val(this.downscaleSliderValue(this.sliderGyroFilterMultiplier));
    }
    this.FilterReset = true;
    this.calculateNewGyroFilters();
`
FC.TUNING_SLIDERS.slider_gyro_filter  is set to 1 instead of 100
and FC.TUNING_SLIDERS.slider_dterm_filter_multiplier
ist not set during reset method 
is this correct?

haslinghuis, 2021-09-17:
Please test with the firmware and configurator development versions mentioned in 2605.

mrRobot62, 2021-09-17:
I closed this issue due to duplicated and reference #2605 - all ongoing checks and results will be published in #2605

closed 2021-09-17 by mrRobot62